ClusterDuck-Protocol/ClusterDuck-Protocol #442: The link for API | Doc not working

tqrahman, 2024-08-29:
To which page do you want to add to the wiki?
readme.md

What do you want to change/add?
We need to change the link to API | Doc or remove it.
[image]

Brijeshthummar02, 2024-10-02:
can you Clarify the issue in detail.

amirna2, 2024-10-02:
@Brijeshthummar02 I think it refers to the badge in the project front-page.
[image]

If you click the API badge it throws a 404 not found error.
It's looking for this link `https://clusterduck-protocol.github.io/ClusterDuck-Protocol/html/index.html`
But not finding it.

This makes sense, because documentation is not (and should not be version controlled).
Previously the HTML pages were generated in the repo and where mistakenly checked in..which is why the link was working before.

But this is no longer the case. Instead the document should be generated  using doxygen and then published as part of the CI pipeline..
You can see the pipelines here:
https://github.com/ClusterDuck-Protocol/ClusterDuck-Protocol/tree/master/.github/workflows

`doxygen.yml` generates the docs and publishes them in the `docs` folder.

Brijeshthummar02, 2024-10-02:
so should i remove the link or it needs to be updated, if updated provide me the correct link.

amirna2, 2024-10-02:
Well, it's not that easy because the link is actually correct.
If you go to the action page you can see the pipeline runs.
https://github.com/ClusterDuck-Protocol/ClusterDuck-Protocol/actions

But I would say this is not a "good first issue" and should probably be addressed by a maintainer as it may require changes to the repository settings. @faradaym

amirna2, 2024-10-02:
@Brijeshthummar02 Actually what seems to be happening is that all the doxygen files are published in the `gh-pages` branch. Right here: https://github.com/ClusterDuck-Protocol/ClusterDuck-Protocol/tree/gh-pages

You can see all the pages are generated in the root directory..including the `index.html`
But if you look at the `docs` folder in the master branch of the repo: 
https://github.com/ClusterDuck-Protocol/ClusterDuck-Protocol/blob/master/docs/index.html

You see there is a custom `index.html` that redirects to `html/index.html`

The assumption is that the deployement to `gh-pages` wouls create an `html` folder in which the actual doxygen generated `index.html` page would be present..

But for some reasons the deployment is done all directly in the root folder.

So that's the issue. It's either in the `doxyfile` or in the doxygen.yml` pipeline.

Brijeshthummar02, 2024-10-02:
yea exactly that's what i feel, also it should not be labeled good first issue, as it's complicated.

amirna2, 2024-10-02:
@Brijeshthummar02 @faradaym The issue was indeed a github settings.

Github was told to build from `master/docs` which then caused it to pick up the index.html that had the re-direct to an `html` folder that doesn't exist. Instead it should be told to build and deploy from `gh-pages` and from the `/ (root)` folder.

I updated the settings and now you can access the docs from this link:
https://clusterduck-protocol.github.io/ClusterDuck-Protocol/index.html

That said we still need to update the link in `API` badge in the readme:
https://github.com/ClusterDuck-Protocol/ClusterDuck-Protocol/blob/master/README.md

@Brijeshthummar02 would you mind doing this?

Also there is a secondary issue with the page itself. The logo is not displayed correctly. 
I think this is because the logo images are not copied to the gh-pages. Probably can open a new ticket for that.

TimoWielink, 2024-10-02:
I was actually working on this yesterday, we need to update the GH actions, since we don't want to store the HTML inside the codebase for doxygen the HTML in the GH pages is old and not up to date. I was working on a fix on my fork. Will create a new issue for Doxygen in general. and we can fix the link

faradaym, 2025-02-04:
Link was removed in #470

closed 2025-02-04 by faradaym