ClusterDuck-Protocol/ClusterDuck-Protocol #446: Publish fails when message too long

tqrahman, 2024-09-23:
**Describe the bug**
A CDP packet can only hold 256 bytes. The data portion (the part where we can add the message) can only hold 229 bytes. If it is more than that, the packet will not publish. However there is no indication to the user why it doesn't fail. 

It happened twice two me where I couldn't figure out why it wasn't publishing, since it was not telling me why. I assumed it was MQTT issues or WiFi.

We need to have an error catcher that tells the user that the data is more than 229 bytes and what the current byte count is, so that they know why it is not publising.

**To reproduce**
Send a packet to a papa duck that is more than 256 bytes long (message is more than 229 bytes). 

**Expected behavior**
The packet will fail to publish without any reason why.

**Additional context (please complete)**
I am using CDP master.

amirna2, 2024-09-23:
Was this a sensor message or a chat?

amirna2, 2024-09-23:
But the bug is very likely here:
[ClusterDuck](https://github.com/ClusterDuck-Protocol/ClusterDuck-Protocol/blob/master/src/Ducks/Duck.cpp#L282-L284)
This method checks for the message length, but doesn't return an error if it's too long!

BrentonPoke, 2024-10-02:
Do we want the error only to be communicated across serial, or also be communicated in a packet?

tqrahman, 2024-10-08:
So this is happening in the PapaDuck. The PapaDuck receives the message from the MamaDuck fine, but then fails to publish, after few attempts, I realized that it was due to the length of the message that the PapaDuck is trying to push through MQTT. Right now it just fails to publish with no reason why.

The question is should we log in the PapaDuck that it is failing because the length of message is too long?

BrentonPoke, 2024-10-18:
> So this is happening in the PapaDuck. The PapaDuck receives the message from the MamaDuck fine, but then fails to publish, after few attempts, I realized that it was due to the length of the message [...]

I'm not sure where it would log to. That could be something that we allow someone to pass an address to. Or maybe create a method that takes a lambda for handling errors.